OHF-Voice/micro-wake-word #16: BL606P Chip

Jab2870, 2024-03-18:
Hi,

Not sure if this is at all feasible, but is there any chance of getting microwakeword running on the BL606P chip?

The reason I ask is that pine64 have recently announced their PineVox, a "smart speaker" designed to be used locally. This is obviously likely to be appealing to the same sort of people who are interested in microwakeword. 

https://pine64.org/documentation/PineVox/

kahrendt, 2024-03-18:
That's an interesting device! It seems that it would be possible to run inference on the BL606P chip, as the SDK has [support for TensorFlow Lite Micro](https://github.com/bouffalolab/bouffalo_sdk/tree/master/components/ai/TensorFlowLite). It's unclear to me, either way, if the chip is fast enough to run inferences that keep up with the incoming audio. I don't have any experience developing on that platform and do not have any of the hardware to work with, so I won't be able to help directly with porting the code. The [ESPHome component](https://github.com/esphome/esphome/tree/dev/esphome/components/micro_wake_word) should be a good starting point if someone want's to try to write it themselves though!

gamelaster, 2024-11-16:
Hi,

BL606P have high performance RISC-V core (T-Head C906) solely dedicated for algorithms.
It should be fast enough to run microWakeWord without acceleration.

BL606P should have also neural network accelerator, but it still need some testing to figure out if it will be possible to integrate it with TensorFlow Lite Micro, so will see.

kahrendt, 2024-12-18:
I'd love to hear from anyone who tries to make this work on different microcontrollers! At this time, I don't plan on personally writing the on device inference code for new platforms.

closed 2024-12-18 by kahrendt

gamelaster, 2024-12-18:
@kahrendt I definitely plan to port this to BL606P, with CPU/software-based TensorFlow Lite Micro (at least for beginning).
I should get to this in upcoming weeks.