OHF-Voice/micro-wake-word #19: Multiple wake words possible?

DieKatzchen, 2024-04-24:
Would it be possible to have multiple wake words on the same device?

kahrendt, 2024-04-24:
Yes, it is possible with two different approaches:
1) The ``micro_wake_word`` component in ESPHome runs two models simultaneously. I have implemented this, but the code needs some clean up. I'm hoping to submit a PR to ESPHome in the next week. You should be able to run two of the current models at the same time, but then the ESP32-S3 is basically at max capacity in terms of CPU usage.
2) Train a single model with multiple wake word probability outputs. I also have this roughly working, but I haven't achieved results that are comparable to single word models (which is to be expected). I want to explore this more in the future with a different model architecture, but it requires some significant changes to the current training process. Currently, all the metrics and related calculations assume a single output indicating the wake word or not.

DieKatzchen, 2024-04-24:
Approach 1 would be sufficient for my current needs, although as you say it is not ideal.
I assume that with approach 2 it would be possible to know _which_ wakeword was triggered?

kahrendt, 2024-04-28:
I've made a [draft PR](https://github.com/esphome/esphome/pull/6653) for the ESPHome component that supports running two models at once, so I'm closing this issue.

closed 2024-04-28 by kahrendt

kahrendt, 2025-04-24:
Adding multiple wake words into a single model decreases accuracy significantly, at least with the same number of parameters. The current MixedNet. Different architectures may be more efficient at it, but it isn't something I've explored heavily. The MixedNet architecture is quite similar to ``ds_tc_resnet`` in a lot of ways, where it mainly uses depthwise convolutions. If you only provide 1 kernel size per layer with the ``mixconv_kernel_sizes`` parameter, then it will be exactly a single depthwise convolution.

Training multiple wake words in a single model requires quite a bit of refactoring since each validation batch assumes there is a binary output when computing the estimated false accepts per hour.