OHF-Voice/micro-wake-word #21: how to infer the streaming

kismeter, 2024-05-31:
trying to infer whole "Alexa.wav" audio to alexa.tflite using predict_clip, it can be detected the keyword. but when using streaming audio (get 480 samples from "Alexa.wav" with 320 steps) predict_clip result show keyword not detected

kahrendt, 2024-05-31:
The ``predict_clip`` function expects the entire clip to be given at once. It uses TensorFlow's microfrontend Python op to generate the spectrogram features, and that op requires the entire clip at once (it has auto gain control and noise reduction, which requires memory of previous audio windows).

If you want to manually stride the audio window like in your code, you need to use a different feature generation approach. The easiest way is to use the ``tflite_micro`` runtime with the preprocessor model. [This notebook from alior101's repo](https://github.com/alior101/Custom-MicroWakeWord-Generator/blob/main/04_testing_notebook.ipynb) has a great example showing how to do this.

kismeter, 2024-06-03:
so the streaming infer must use tflite_micro runtime with preprocessor model?

kahrendt, 2024-06-03:
Yes, that is the most straightforward way to generate the spectrogram features for streaming audio. I haven't personally experimented with it as the ``tflite_micro`` runtime doesn't run on my Mac's M1 processor, but alior101 has had success with that approach.

kismeter, 2024-06-03:
also not working on Mac's M2. Thanks @kahrendt

closed 2024-06-03 by kismeter