OHF-Voice/micro-wake-word #26: ESPHome does not git pull updated model during re-compile

Omnipius, 2024-06-25:
In trying to experiment with a handful of candidate models and probability cutoffs, I noticed that ESPHome wasn't pulling an update from my model GitHub repo. Rather, it only seems to use the last model and Jason downloaded from the exact URL provided, even going back several merges to main. I'm not sure if that's a microwakeword issue or an ESPHome issue.

Additionally, it would seem that the ESPHome firmware for the ESP32-S3-BOX-3 does not pass through yaml substitutions of probability_cutoff and sliding_window_average_size like it does model. Is there a direct way to substitute these?

jamesmyatt, 2024-06-26:
Are you doing this via `external_components`? If so, you need to set the `refresh` parameter to force an update.

Omnipius, 2024-06-26:
I'm pulling in:

```
packages:
  esphome.voice-assistant: github://esphome/firmware/wake-word-voice-assistant/esp32-s3-box-3.yaml@main
```

That yaml uses the micro_wake_word component of ESPHome directly. So the issue seems to be inside there.

kahrendt, 2024-06-27:
I'm not sure what is going on with ESPHome and GitHub. I've been encountering this all day where ESPHome fails to download something from GitHub unless I retry many times. It's not just limited to to mWW models, I'm having issues whenever it tries to install a PlatformIO library, refresh an external component, or download an image (like for the S3 boxes), so I don't believe it is particular to microWakeWord.

The easiest way to adjust the probability_cutoff and sliding_window_average_size would be to copy and paste the entire yaml file into a new device in the ESPHome dashboard. You can then adjust those parameters directly in the copied yaml at the appropriate spot.

jamesmyatt, 2024-06-28:
there's a "refresh" option for remote packages too: https://esphome.io/guides/configuration-types.html#remote-git-packages

jamesmyatt, 2024-06-28:
Also you can override substitutions when including a file via the `vars` option, e.g. https://esphome.io/guides/configuration-types.html#packages-as-templates

jamesmyatt, 2024-06-28:
But I think this is an ESPHome question. Not a microWakeWord one.

kahrendt, 2024-07-03:
I haven't been having issues the last few days with ESPHome pulling things from GitHub. If you are still encountering this, please open an issue in the ESPHome issues repository.

closed 2024-07-03 by kahrendt