OHF-Voice/micro-wake-word #28: Is this repository more accurate then OpenWakeWord ?

dilerbatu, 2024-07-16:
I want to learn that what is the difference between openwakeword and this repository ? Does accuracy differ ?

s-dome17, 2024-07-17:
This is smaller model which can run on low power devices such as ESP32 S3

synesthesiam, 2024-07-18:
openWakeWord uses a pre-trained model from Google to increase accuracy. This works great, but the Google model is pretty complex.

microWakeWord is smaller and trained from scratch so it can run on ESP devices.

s-dome17, 2024-07-18:
@synesthesiam I was trying to train a wake word using microWakeWord for Hey Pixa, but was finding it a bit difficult compared to openWakeWord, have you trained any wakeword here before? something you can help with?

kahrendt, 2024-07-18:
It is much easier to train a good model with openWakeWord than microWakeWord. While the actual training of a mWW model only takes a couple hours on my hardware, it usually takes me a couple weeks of tweaking the TTS samples to get a usable model. That is a big part of the challenge for writing an all in one training script for microWakeWord!

It is hard to compare accuracy for wake word models as there isn't a standardized test that describes all situations well. For my specific tests, I have found the new V2 models are slightly better than openWakeWords, but it is unclear if this is reflected in real world use.

dilerbatu, 2024-07-18:
Thanks for information also for the work :rocket:

gustvao, 2024-07-22:
@kahrendt i have been able to use the pre-trained words efficiently but funny enough when my wife says alexa or hey jarvis from the same distance/volume than me it does not get recognized..

i tried using a higher tone and microwakeword stop detecting..

do you thing we might be training the model only with male samples? what about having always a male and a female option? this way it would be more inclusive given that we can use more than 1 wakeword at a time in the same device :)

amazing work and thanks again

kahrendt, 2024-07-22:
Interesting result! The samples are all generated using models trained on the LibriTTS dataset. The dataset is roughly balanced between genders as a whole, so in theory, the generated samples should also be balance. I did however restrict the number of voices when generating samples for several of the V2 models, and it is possible that resulted in an unbalanced set. I'll investigate this, thanks for pointing it out!

Out of curiosity, have you tried the "Hey Mycroft" wake word? I'd be interested to hear if you experience the same behavior, as I used many real recordings in addition to TTS generated samples.

gustvao, 2024-07-22:
hey Kevin! interesting enough, hey mycroft works with her :)

kahrendt, 2024-07-22:
I had another thought as well that could improve it. The ESPHome 2024.7.0 release changed how loud the microphone audio was. Basically, it was 4 times quieter than previous version. The 2024.7.1 release reverted that change, so it should match the same behavior as before. If you were on the 2024.7.0 release, I'd appreciate if you updated ESPHome and tested again!

gustvao, 2024-07-22:
hey kevin, everything was on 7.1 already ;) hey microft works, all other v2
models do not for her: ok nabu, hey jarvis nor alexa

TheStigh, 2024-08-27:
@kahrendt Hi, I experience the same with Alexa and Hey Jarvis, my wife is not picked up at all unless she really lower the pitch of her voice, resulting in her telling me to remove it from the livingroom :) Currently running 2024.8.0. Tested on both the Atom Echo and the ESP32-S3-BOX-3.

kahrendt, 2024-08-27:
> @kahrendt Hi, I experience the same with Alexa and Hey Jarvis, my wife is not picked up at all unless she really lower the pitch of her voice, resulting in her telling me to remove it from the livingroom :) Currently running 2024.8.0. Tested on both the Atom Echo and the ESP32-S3-BOX-3.

Thanks for the feedback. I'm brainstorming ideas on how to properly address this, as the bias only gets worse when I additionally train on collected real samples. One thought I is to oversample the TTS samples that are based on female voices to see if that improves things. I'm also suspicious that the samples generated with slerp (so 2 trained voices are mixed together) are making this worse.

TheStigh, 2024-08-28:
> Thanks for the feedback. I'm brainstorming ideas on how to properly address this, as the bias only gets worse when I additionally train on collected real samples. One thought I is to oversample the [...]

I'm happy to test and use my wife as the female tester :)

thirstyone, 2024-10-01:
can it be run on esp32c3 by any chance?

Digital-Ark, 2024-10-25:
This is very cool.  I successfully got an ESP32-S3 responding to "Hey Jarvis".

Unfortunately, I wasted a lot of time training a new "Hey Eddie" openWakeWord, but I now realize this microWakeWord is a separate thing.

On device wake words are so much better.  No continuous bandwidth to Home Assistant, snappier response.

Are you in any way soliciting for new wake words?  My plan was to make my Assist act like Eddie the Shipboard Computer from The Hitchhiker's Guide to the Galaxy.  The AI prompt is working well, and the assistant is endlessly cheerful, and I even have a reasonably okay MaryTTS voice effect with reverb.

kahrendt, 2024-10-26:
I'm not actively working on new wake words at this time. It would be better to build a decent all in one training script prepared than me attempting to do more wake words. The biggest problem is it takes a lot of tweaking and experimentation to get a usable model, especially with TTS samples. Lately I've been focused on helping getting Nabu Casa's voice satellite firmware ready to go, so getting a script together has been a lower priority. As I get the firmware components merged into ESPHome, I'll spend time preparing a script to make it easier for everyone to experiment with.

Digital-Ark, 2024-10-26:
I understand, thank you for all you've already done on this, it's absolutely amazing!

yf-fan-org, 2024-12-24:
@kahrendt  Can this pre-trained model run on a normal Ubuntu server? My test outputs are all 0

erkamkavak, 2024-12-25:
> I'm not actively working on new wake words at this time. It would be better to build a decent all in one training script prepared than me attempting to do more wake words. The biggest problem is it [...]

Can you explain why it takes a lot of time to tweak tts samples? Is this because tts generation is not really successful and variation is not enough? 
I wonder if using new models like openai's gpt 4o mini audio or google's gemini flash 2.0 can be used to generate tts samples since you can customize the tone or pronunciation in the output audio. If the biggest problem of microwakeword is dataset generation, maybe these models can solve it

StuartIanNaylor, 2024-12-28:
>> I wonder if using new models like openai's gpt 4o mini audio or google's gemini flash 2.0 can be used to generate tts samples since you can customize the tone or pronunciation in the output audio. If the biggest problem of microwakeword is dataset generation, maybe these models can solve it

There are also later SotA TTS that also can clone voices from short samples https://github.com/fishaudio/fish-speech and https://github.com/Plachtaa/VALL-E-X
You are right though as I listened through to the 1000 TTS wav samples Piper makes and its 2 very close voices when being slightly more make with a little pitch variration.
Its a million miles away from the intonation variance differing human voices make from various regions...
Its also synthetic sounding and likely that is embedded into the spectra and with the old saying 'garbage in, garbage out' that dataset makes quite bad models.

erkamkavak, 2024-12-29:
> I wonder if using new models like openai's gpt 4o mini audio or google's gemini flash 2.0 can be used to generate tts samples since you can customize the tone or pronunciation in the output audio. If the biggest problem of microwakeword is dataset generation, maybe these models can solve it

By the way I tested this approach and I have got a pretty good result(I have previously trained openwakeword model with piper samples and compared to it this is much better). I have generated about 10000 samples with gpt4o mini api and verified them with gemini api.

kahrendt, 2024-12-29:
> Can you explain why it takes a lot of time to tweak tts samples? Is this because tts generation is not really successful and variation is not enough? I wonder if using new models like openai's gpt [...]

Correct, the variation does not seem to be enough. When I mention tweaking generation settings, I do mean playing with the various noise settings and try to use slightly different phonetic pronunciations to increase the variety.

Other TTS models could be very useful, but it's possible that using samples generated with commercial TTS engines are not licensed for training a new model with. That's probably not a large concern if you are only using it yourself, but it could potentially be an issue if you want to share the model with others.

kahrendt, 2024-12-29:
> It gets even stranger later where 3D room RIR's are converted into wavs without use of something like https://github.com/LCAV/pyroomacoustics or https://github.com/DavidDiazGuerra/gpuRIR to create the point sources of mic and speaker to record accurately how that room RIR would record the sample played... I guess they don't understand what RIR's are as the same but from a different angle is happening here. [OHF-Voice/wake-word-collective#11](https://github.com/OHF-Voice/wake-word-collective/issues/11)

In an ideal world, collected wake word samples would be recorded with proper audio gear in a studio where we just get the high quality sample. We then could augment them freely with various RIRs to simulate many different rooms. We could also easily add a different source of noise to the sample coming from a different spot in the same simulated room. However, collecting samples in this way is expensive and challenging. Most people aren't equipped to record samples in this way, so it would be difficult to get the variety necessary to train a robust model. Collecting as many samples as possible, which capture real room impulse responses, is a simpler alternative that can still train a decent performing model.

StuartIanNaylor, 2024-12-29:
The best collected wake word samples are from the device of use.
It is possible to create a rolling window to capture the KW on hit and send over websockets to a middle server to collate a local dataset.
It would be great if we had that function and an opt-in to upload to a opensource dataset with metadata filled in by the user.
Same with ASR as there is a pretty clear logic if a command is correct and those sentences of device would also be great.

It would be great also if you could load firmware or have a state machine where the device becomes a microphone so that you can use it for recordings.
https://github.com/StuartIanNaylor/ProjectEars/tree/main/dataset/reader
Is just a CLI recording 'boutique' that prompts the user to say keywords and words from phonetic pangrams 'sentences' containing all phonemes in a language.

The whole point of not just having a singular microphone to capture farfield is that the RIR causes the ripple of audio to hit and reflect off surfaces, all at different distances and then returning and mix at the microphone.
This depending on mic to source disyance causes huge spectra distortions as differing room time delays mix at the microphone.
That is why near all farfield tech tries to focus and remove the effects of RIRs before submitting to KW and ASR. Its RIRS and there effect why we have all these various multi microphone technologies in various voice tech all with differing algs.
You don't capture RIRs in a wav just the distortion the mix of reverbervation will of caused in that room causing a big increase in entropy of captured samples.
Capturing samples on any device of people in any room at any distance, with zero meta data will create a relatively useless data set and a pretty pointless alternative.
Even the device you are using processes and removes RIRs by extracting voice via 2 microphones and cleans the sample before the KWS.
So why would you train that KWS on samples containing farfield reverberation?

If you could just record room impulse responses via a simple voice sample the whole industry would need no microphone tech or algs, just the datasets of capture and the corresponding models.
Good luck with the models you will create.

StuartIanNaylor, 2025-01-08:
I had a go in the esphome to see if any guru's would enable the usb audio class drivers on the xmos so I can plug in and use as a microphone.
I haven't really bothered having a go at creating a dataset as really need a xmos device and one I can play with python for recording and having a listen and try to analyse how good the farfield is.
I am sort of blind without that and the request seems very closed to esp32-s3 only likely to its detriment.

The samples from Piper are far too similar and without enough variance which is an easy fix by with newer TTS running on something with a GPU even my Xeon workstation with a RTX3050 can run reasonable TTS.
The recorded RIRs is just total bunkum as they are all recorded at 1.5m but only a few are typical room sizes whilst the dataset contains many huge halls, churches and even forrests which why that dataset or that method was chosen is a complete mystery.
When you have speech enhancement for farfield then your spectra is going to be different to a single mic, but without audacity and some spectra and some analysis and trial and error I just don't know.
Not much you can do about Mlcommons as there is a huge amount of bad but analysing in a database and grabbing an even spread of phonetics by sylable will help.
Putting any voice in the noise dataset increases cross entropy as a classification model is not AI just a pure graph and it will lower overall KW hit probablity. The 'cocktail party' problem is equal to deverberation and the idea you can just add to a classification model is a basic 101 missunderstanding of the huge amount of research and papers published on how to tackle this.
Music can be very similar any vocals are so near voice the same is true and often certain instruments can give very vocal looking spectra.
You can check this by using your dataset as the input to a trained model and checking the probability of any label.
Often I would do that to remove the worst 5-10% of a dataset and retrain.

StuartIanNaylor, 2025-01-13:
This seems an easy way for KW.
`pip install coqui-tts`
```
from TTS.api import TTS

tts = TTS("tts_models/multilingual/multi-dataset/xtts_v2", gpu=True)

languages = tts.languages
speakers = tts.speakers
for speaker in speakers:
    for lang in languages:
    # generate speech by cloning a voice using default settings
        if lang != "hi" and lang != "ko":
            tts.tts_to_file(text="Hey! Jarvis!",
                            file_path=speaker + "-1-" + lang + ".wav",
                            speaker=speaker,
                            language=lang,
                            split_sentences=False
                            )
            tts.tts_to_file(text="Hey Jarvis",
                            file_path=speaker + "2-" + lang + ".wav",
                            speaker=speaker,
                            language=lang,
                            split_sentences=False
                            ) 
```
 That will end you up with 1700+ 'Hey Jarvis' with a bit more variation and is a simple script if you battle pip for the missing modules.

StuartIanNaylor, 2025-02-09:
> I am not an expert in AI and speech recognition at all, but as it seems to me. if you record at least 100-200 words and spend time from a real device in different rooms of your house the result will be better than what I hear was obtained on a “robotic” piper.

If you augment the 100-200 words to a dataset of say 2000-4000 samples in each label you can get a very accurate KWS, but unfortunately it will only work for yourself.
You can also do on-device training but the on-device would likely need to be a middleware N100 or Casa Cloud, where you ship out a model OTA to the microcontroller.
I am sort of thinking that a PiZero2 or equiv has so much more compute than a ESP32 that its a shame they have been ommited.

2000-40000 samples in each label is really a tiny fraction of what the correctly named Big Data of Google would use.
https://developers.google.com/machine-learning/crash-course/overfitting/data-characteristics gives a rough guide to dataset sizes with KWS parameters often being arounf 400k.

> I could be wrong about something or everything, but for me, as a non-native English speaker, it's quite agonizing to talk to the assistant (to wake him up). As for recognizing commands, I use the big VOSK model, it works fine for me. So these are just thoughts out loud.

Its Whisper as SotA WER it publishes are for a few languages on the large model with a 30 second context.
https://www.speechly.com/blog/analyzing-open-ais-whisper-asr-models-word-error-rates-across-languages
Use certain languages, with certain model sizes with short context and the WER is likely the worst where its debatable if your language is supported.

Both Tensorflow and Torch have on-device training methods, but just collating and retraining can also be done if you have the patience as the idle time of a smart speaker is usually very large.

I did a CLI Word Boutique to collect word samples from spoken KW and !KW via what are phonetic pangrams which is a nonsense sentence containing all main Phonemes of a language. They can be created for any language so a few sentences can be the basis of adding 'own voice' to any dataset.
Some rough hacks are in https://github.com/StuartIanNaylor/Dataset-builder or https://github.com/StuartIanNaylor/ProjectEars/tree/main/dataset/reader

Synthetic wise https://github.com/coqui-ai/STT ⓍTTSv2 https://github.com/coqui-ai/TTS and https://github.com/netease-youdao/EmotiVoice are very good and why existing needs to wait until its refactored and rebranded as HA before being used is questionable as there is an ever changing slection of great opensource avail Vosk being one.

Even for a KWS accuracy likely could be improved by tayloring to a language family such as https://github.com/AI4Bharat/Indic-TTS as there are likely clear language branches that do not need a specific language KWS https://en.wikipedia.org/wiki/Indo-European_languages#/media/File:Indo-European_Language_Family_Branches_in_Eurasia.png

If you check my repos I have been having a go again with synthetic datasets as with simple models accuracy is the dataset and can be quite a bit of work. Providing datasets for regional branches is likely best done by native speakers of that native branch and maybe hack some of the scripts and methods I have used.
https://github.com/StuartIanNaylor/create-dataset
https://github.com/StuartIanNaylor/syn-dataset
You can test out with a KWS script of your choice as I use https://github.com/google-research/google-research/tree/master/kws_streaming but having to branch at a commit of some years ago.
But use https://github.com/StuartIanNaylor/tf-kws to test for now.

ab-tools, 2025-11-09:
> I do think its a shame either the USB Class1/2 Audio drivers or I2S drivers on the Pi have not been provided.
Likely https://github.com/tRackIT-Systems/snd-i2s_rpi the dumb slave I2S microphone drivers for the Pi would work that Adafruit document. Always wondered though if the 24/16bit data is being present as a 32bit pcm as hence why they always seem so quiet.

At least for me USB UAC Audio works well using a [XMOS microphone array over USB](https://www.ab-smarthouse.com/en/product/mikrofon-array-xmos-aec/) connected to a Raspberry Pi Zero 2 using openWakeWord.

But I'm currently looking into possibly using a ESP32-P4 instead of the Raspberry Pi Zero 2 - then with microWakeWord instead, of course.