OHF-Voice/micro-wake-word #30: Custom wake word tutorial

khaue, 2024-09-18:
It should be possible to create a more detailed tutorial on how to create new custom wake words? Perhaps a YouTube video?

00xBAD, 2024-11-21:
A Google Colab like [OpenWakeWord](https://colab.research.google.com/drive/1q1oe2zOyZp7UsB3jJiQ1IFn8z5YfjwEb?usp=sharing#scrollTo=1cbqBebHXjFD) should be enough for novice users.

zigomatichub, 2024-12-11:
Hello,
I want to have a wakeword in French basicaly to use with openwakework add-on of Home assistant or directly in ESP Home.

Then I tried and successfully created a .tflite with the help of https://studio.edgeimpulse.com/ by using deployment method "Think Silicon Neox GA100".

But it seems that is not working or may be there are mutiple kind of data that is not accepted by the add-on.

I'm still searching.

craig0r, 2024-12-18:
> A Google Colab like [OpenWakeWord](https://colab.research.google.com/drive/1q1oe2zOyZp7UsB3jJiQ1IFn8z5YfjwEb?usp=sharing#scrollTo=1cbqBebHXjFD) should be enough for novice users.

I may be mistaken, but I think OpenWakeword and MicroWakeword training are two different processes. (My uncertainty on this speaks to the broader topic that better, easily available documentation would benefit everyone.)

kahrendt, 2024-12-18:
If you are comfortable setting up Python virtual environments and generally hacking around with this stuff, please try the new training notebook! A lot more can be done to simplify the process, but please experiment and let me know whats working and what isn't!

closed 2024-12-18 by kahrendt