OHF-Voice/micro-wake-word #34: microWakeWord running natively on RP2350 or RP2040 microcontroller platforms?

Hedda, 2024-10-25:
Can the microWakeWord engine run natively on RP2350 or RP2040 microcontroller platforms (

* https://en.wikipedia.org/wiki/RP2350
  * https://www.raspberrypi.com/products/rp2350/
    * https://www.raspberrypi.com/products/raspberry-pi-pico-2/

and/or

* https://en.wikipedia.org/wiki/RP2040
  * https://www.raspberrypi.com/products/rp2040/
    * https://www.raspberrypi.com/products/raspberry-pi-pico/

There looks to be a few unofficial ports of the [TensorFlow Lite Micro library](https://www.tensorflow.org/lite/microcontrollers) for the RP2040 but obviously RP2350 should perform better. Raspberry Pi posted this blog post when they added RP2040 support, then silence:

* https://www.raspberrypi.com/news/a-big-bang-update-for-tensorflow-lite-for-microcontrollers/

* * https://github.com/raspberrypi/pico-tflmicro

Note that there is also not yet an official Raspberry Pi Pico 2 board with Wi-Fi but there are third-party boards with WiFi and LAN, see:

* https://www.raspberrypi.com/news/our-rp2350-partners-made-all-this-excellent-stuff-for-you/
  * https://www.raspberrypi.com/for-industry/powered-by/product-catalogue/?category=RP2350
    * https://ilabs.se/?s=RP2350&post_type=product
    * https://docs.wiznet.io/Product/iEthernet/W6100/w6100-evb-pico2

RP2350 is newly released so there is not yet widespread support but  RP2040 based development boards are already very popular in schools and STEM education so think they are interesting as alternatives to ESP32 because documentation is often easier as often targeting beginners which in turn means can lead to broader use and use cases.

kahrendt, 2024-10-26:
It's certainly possible if TFLM is available and the models run fast enough on that hardware. I have not worked with the RP boards, so I am not familiar enough beyond that. It would be great if people could build out support on other platforms!

kahrendt, 2024-12-18:
I'd love to hear from anyone who tries to make this work on different microcontrollers! At this time, I don't plan on writing the on device inference code for new platforms.

closed 2024-12-18 by kahrendt

Hedda, 2024-12-30:
Not tested but FYI, they have upgraded the pico-tflmicro project to latest SDK so it should now work with the RP2350 as well:

- https://github.com/raspberrypi/pico-tflmicro