OHF-Voice/micro-wake-word #43: hey_jarvis.tflite model keeps outputting 0

yf-fan-org, 2024-12-24:
https://github.com/esphome/micro-wake-word-models/blob/main/models/v2/hey_jarvis.tflite 
hey_jarvis.tflite model always outputs 0, I would like to ask this model why I input random numbers but the output is always 0, is this model able to work properly?

kahrendt, 2024-12-24:
The model expects a spectrogram like input. Using just random numbers most likely has an input not anywhere close to a spectrogram of the wake phrase, so the model is accurately outputting 0 in this situation. You can use TensorFlow (see the ``microwakeword/audio`` folder in this repo) or [PyMicro-Features](https://github.com/rhasspy/pymicro-features) to generate the appropriate spectrograms from an actual audio clip.

closed 2024-12-28 by kahrendt

mahmoud-othmane, 2025-01-08:
Hi @kahrendt 
Great work overall, and thank you for sharing the code and the models!
I've been trying to make use of a the model using a simple Python script, but the output of the prediction is always zero.
I'm new to this field, so I hope you'll excuse my lack of experience.
Below is the script I was using, so would you help me a bit figure out what's wrong? 
I used ChatGPT, but it does not seem to be working as expected.
Thank you!
```Python
import numpy as np
import tensorflow as tf
import librosa
import sounddevice as sd
from collections import deque

# Path to the .tflite model
MODEL_PATH = "alexa.tflite"

# Parameters
SAMPLE_RATE = 16000  # Audio sample rate in Hz
WINDOW_DURATION_MS = 30  # Window size (30 ms)
STRIDE_DURATION_MS = 10  # Stride size (10 ms)
N_FEATURES = 40          # Number of spectrogram features
CHUNK_DURATION_MS = 100  # Duration of audio chunk to process (100 ms)
CHUNK_SIZE = int(SAMPLE_RATE * CHUNK_DURATION_MS / 1000)  # Number of samples per chunk
WINDOW_SIZE = int(SAMPLE_RATE * WINDOW_DURATION_MS / 1000)  # Samples in a window
STRIDE_SIZE = int(SAMPLE_RATE * STRIDE_DURATION_MS / 1000)  # Samples in a stride
FEATURE_QUEUE_SIZE = 3  # Number of spectrogram slices for each inference window
SLIDING_WINDOW_SIZE = 5  # Sliding window size for detection confirmation
PROBABILITY_CUTOFF = 0.9  # Probability threshold for detection

# Initialize a sliding window buffer for audio data
audio_buffer = deque(maxlen=CHUNK_SIZE)

# Initialize a sliding window buffer for spectrogram features
feature_buffer = deque(maxlen=FEATURE_QUEUE_SIZE)

# Add sliding window to track probabilities
sliding_window = deque(maxlen=SLIDING_WINDOW_SIZE)

# Helper function: Generate spectrogram features
def extract_spectrogram(audio, sample_rate):
    hop_length = int(STRIDE_DURATION_MS * sample_rate / 1000)
    win_length = int(WINDOW_DURATION_MS * sample_rate / 1000)
    spectrogram = librosa.feature.melspectrogram(
        y=audio,
        sr=sample_rate,
        n_fft=win_length,
        hop_length=hop_length,
        n_mels=N_FEATURES,
        power=2.0
    )
    log_spectrogram = librosa.power_to_db(spectrogram)
    return log_spectrogram.T  # Transpose for time-major shape

# Load the TFLite model
interpreter = tf.lite.Interpreter(model_path=MODEL_PATH)
interpreter.allocate_tensors()

# Get input and output details
input_details = interpreter.get_input_details()
output_details = interpreter.get_output_details()
input_shape = input_details[0]['shape']
window_size = input_shape[1]  # Expecting 3 slices of 40 features
print("Input tensor shape:", input_shape)

# Function to process audio and run inference
def process_audio(audio_chunk):
    global feature_buffer, sliding_window

    # Convert audio to spectrogram features
    features = extract_spectrogram(audio_chunk, SAMPLE_RATE)

    # Add the newest slice of features to the feature buffer
    for feature_slice in features:
        feature_buffer.append(feature_slice)
        if len(feature_buffer) >= FEATURE_QUEUE_SIZE:  # If the buffer is full
            # Prepare input for the model
            model_input = np.array(feature_buffer, dtype=np.float32)
            model_input = np.expand_dims(model_input, axis=0).astype(np.int8)  # Add batch dimension
            
            # Run inference
            interpreter.set_tensor(input_details[0]['index'], model_input)
            interpreter.invoke()
            output = interpreter.get_tensor(output_details[0]['index'])
            probability = output[0][0]

            print("Probability:", probability)

            # Add probability to sliding window
            sliding_window.append(probability)

            # Check if wake word is consistently detected
            if sum(p > PROBABILITY_CUTOFF for p in sliding_window) >= 3:  # Majority of the last 5 windows
                print("Wake word detected! Probability:", probability)

# Real-time audio callback
def audio_callback(indata, frames, time, status):
    global audio_buffer

    if status:
        print("Audio Status:", status)

    # Flatten audio input and append to buffer
    audio_data = indata[:, 0]  # Use the first channel if stereo
    audio_buffer.extend(audio_data)

    # Process audio in chunks
    if len(audio_buffer) >= CHUNK_SIZE:
        audio_chunk = np.array(audio_buffer, dtype=np.float32)
        process_audio(audio_chunk)

# Start real-time audio stream
print("Listening for wake word...")
with sd.InputStream(
    channels=1,
    samplerate=SAMPLE_RATE,
    callback=audio_callback,
    blocksize=CHUNK_SIZE
):
    try:
        while True:
            pass  # Keep the script running
    except KeyboardInterrupt:
        print("\nStopped listening.")
        
```