OHF-Voice/micro-wake-word #46: custom models crashing

Kelerion-UK, 2024-12-26:
I'm pretty sure I've done this before with mWW.. (I've definitely done it with oWW)... am trying to get a custom wakeword built (just for learning).. and I've managed to get the environment built (not as easy as it could be! lol).. and have buily a tflite.. I've added to the yaml build but its crashing on loading and rebooting the device (both s3box and pve).. I'm getting this error:

`[23:40:29]Didn't find op for builtin opcode 'DEQUANTIZE'
[23:40:29]Failed to get registration from op code DEQUANTIZE
[23:40:29] 
[23:40:29][E][micro_wake_word:058][inference]: Failed to allocate tensors for the streaming model
[23:40:29]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[23:40:29]
[23:40:29]Core  0 register dump:
[23:40:29]PC      : 0x4203613a  PS      : 0x00060830  A0      : 0x82035905  A1      : 0x3fccd240  
WARNING Decoded 0x4203613a: tflite::MicroInterpreterGraph::FreeSubgraphs() at /home/craig/Development/Dave/.esphome/build/vpe/components/esp-tflite-micro/tensorflow/lite/micro/micro_interpreter_graph.cc:157 (discriminator 1)
[23:40:29]A2      : 0x3fcd0dc0  A3      : 0x0000000d  A4      : 0x00000000  A5      : 0x00000000  
[23:40:29]A6      : 0x00000049  A7      : 0x00000000  A8      : 0x3c284458  A9      : 0x00000002  
[23:40:29]A10     : 0x00000049  A11     : 0x00000008  A12     : 0x3c287628  A13     : 0x3c0fc888  
[23:40:29]A14     : 0x00000004  A15     : 0x00000014  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
[23:40:29]EXCVADDR: 0x00000006  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  
[23:40:29]
[23:40:29]
[23:40:29]Backtrace: 0x42036137:0x3fccd240 0x42035902:0x3fccd260 0x42013232:0x3fccd280 0x4201324e:0x3fccd2a0 0x42012aca:0x3fccd2c0 0x42013015:0x3fccd2e0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x42036137: tflite::MicroInterpreterGraph::FreeSubgraphs() at /home/craig/Development/Dave/.esphome/build/vpe/components/esp-tflite-micro/tensorflow/lite/micro/micro_interpreter_graph.cc:152
WARNING Decoded 0x42035902: tflite::MicroInterpreter::~MicroInterpreter() at /home/craig/Development/Dave/.esphome/build/vpe/components/esp-tflite-micro/tensorflow/lite/micro/micro_interpreter.cc:86
WARNING Decoded 0x42013232: std::default_delete<tflite::MicroInterpreter>::operator()(tflite::MicroInterpreter*) const at /home/craig/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/12.2.0/bits/unique_ptr.h:95
 (inlined by) std::default_delete<tflite::MicroInterpreter>::operator()(tflite::MicroInterpreter*) const at /home/craig/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/12.2.0/bits/unique_ptr.h:89
 (inlined by) std::__uniq_ptr_impl<tflite::MicroInterpreter, std::default_delete<tflite::MicroInterpreter> >::reset(tflite::MicroInterpreter*) at /home/craig/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/12.2.0/bits/unique_ptr.h:203
WARNING Decoded 0x4201324e: std::unique_ptr<tflite::MicroInterpreter, std::default_delete<tflite::MicroInterpreter> >::reset(tflite::MicroInterpreter*) at /home/craig/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/12.2.0/bits/unique_ptr.h:501
 (inlined by) esphome::micro_wake_word::StreamingModel::unload_model() at /home/craig/Development/Dave/.esphome/build/vpe/src/esphome/components/micro_wake_word/streaming_model.cpp:94
WARNING Decoded 0x42012aca: esphome::micro_wake_word::MicroWakeWord::unload_models_() at /home/craig/Development/Dave/.esphome/build/vpe/src/esphome/components/micro_wake_word/micro_wake_word.cpp:425

I've tried increasing the "tensor_arena_size" in the json.. but no matter the value it doesn't make any difference.. so am guessing thats not the issue... I'm using the latest notebook (updated a week ago)..  so am wondering if I'm missing something fundamental.. 

Thanks!

erkamkavak, 2024-12-28:
I have the same problem. When I only change model url to okay_nabu v2 url, it works fine. It seems there is a problem with the training code. When I look at the commits there is tensorflow update [commit last week](https://github.com/kahrendt/microWakeWord/pull/36). It may be related to this.

erkamkavak, 2024-12-28:
After looking at the pull requests, I saw this [pr](https://github.com/kahrendt/microWakeWord/pull/39). It seems the problem is fixed but on a different branch(main). I am now testing it on the main branch.

closed 2024-12-28 by kahrendt

Kelerion-UK, 2024-12-28:
Sweet.. that worked.. thanks.

PR #48 (2024-12-28): Use the main branch in the training notebook
Trained models weren't able to be properly quantized to run on the device, which was fixed in #39. However the training notebook still cloned a branch with the bug, so this PR changes it to use the main branch.

Fixes #46, #47 if the model is converted with the new notebook.