OHF-Voice/micro-wake-word #49: Use the tf4micro esp fork not the tensorflow libs?

StuartIanNaylor, 2024-12-29:
https://github.com/espressif/esp-tflite-micro
https://github.com/espressif/esp-nn

Even though it seems the esp-tflite-micro is behind the newly partitioned https://github.com/tensorflow/tflite-micro repo.
I am not sure but unless you link esp-nn you will not use the optimised vector instructions of the S3 

<ul dir="auto" style="box-sizing: border-box; padding-left: 2em; margin-top: 0px; margin-bottom: var(--base-size-16); color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><li style="box-sizing: border-box; margin-top: 0.25em;"><p dir="auto" style="box-sizing: border-box; margin-top: var(--base-size-16); margin-bottom: var(--base-size-16);">Kernelwise performance on ESP32-S3 chip</p><ul dir="auto" style="box-sizing: border-box; padding-left: 2em; margin-top: 0px; margin-bottom: 0px;"><li style="box-sizing: border-box;">Numbers are ticks taken for kernel to execute</li><li style="box-sizing: border-box; margin-top: 0.25em;">Chip config: 240MHz, SPI: QPI 80MHz, Data cache: 64KB</li></ul><markdown-accessiblity-table data-catalyst="" style="box-sizing: border-box; display: block;">
Function | ANSI C | Optimized | Opt Ratio | Data info | Memory
-- | -- | -- | -- | -- | --
elementwise_add | 312327 | 71644 | 4.36 | size = 1615 | External
elementwise_mul | 122046 | 30950 | 3.95 | size = 1615 | External
convolution | 4642259 | 461398 | 10.06 | input(10,10), filter(64x1x1x64), pad(0,0), stride(1,1) | External
convolution | 300032 | 43578 | 6.9 | input(8,8), filter(16x1x1x16), pad(0,0), stride(1,1) | External
convolution | 2106801 | 643689 | 3.27 | input(10,10), filter(64x3x3x3), pad(0,0), stride(1,1) | External
depthwise conv | 1192832 | 191931 | 6.2 | input (18, 18), pad(0,0), stride(1,1) filter: 1x3x3x16 | External
depthwise conv | 1679406 | 366102 | 4.59 | input (12, 12), pad(1,1), stride(1,1) filter: 8x5x5x4 | External
max pool | 485714 | 76747 | 6.33 | input(16,16), filter (1x3x3x16) | Internal
avg pool | 541462 | 160580 | 3.37 | input(16,16), filter (1x3x3x16) | Internal
fully connected | 12290 | 4439 | 2.77 | len: 265, ch = 3 | Internal
prelu (relu6) | 18315 | 1856 | 9.87 | size, 1615 | Internal

</markdown-accessiblity-table></li></ul><div class="markdown-heading" dir="auto" style="box-sizing: border-box; position: relative; color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><br class="Apple-interchange-newline">

kahrendt, 2024-12-29:
The microWakeWord implementation in ESPHome uses the Espressif library which automatically imports the ``esp-nn`` code, so we are already taking advantage of the optimized versions that use the vector instructions. Even the regular ESP32 runs the model faster with their assembly versions compared to disabling the ``esp-nn`` optimizations.

closed 2024-12-29 by kahrendt

StuartIanNaylor, 2024-12-29:
I was thinking you would have to adapt the code they provide as it seems for specific input tensors, which changes in each model.
I am not sure but think the optimised code might be the model that runs in the s3-box