OHF-Voice/micro-wake-word #50: Notebook output model not Working

bibidentuhanoi, 2024-12-30:
Hi,  I'm trying to create a custom model of mine own with the current notebook , every model output not working, i always have the errors message as below, i even tried to recreate the "Okay Computer" model (size 30000) from esphome repo with default seting but nothing works, only that model or model that was created from last  month from this repo and esphome one works. 
`Didn't find op for builtin opcode 'DEQUANTIZE'
Failed to get registration from op code DEQUANTIZE

E (823) micro_wake_word: Failed to allocate tensors for the streaming model
E (833) micro_wake_word: Failed to initialize a wake word model Gideon.
E (833) micro_wake_word: Failed to load the wake word model(s) or allocate buffers
W (843) micro_wake_word: Wake word component has an error. Please check logs`

this is the code i use to convert the model:
```
`import os
def convert_tflite_to_h(tflite_file):
    base_name = os.path.splitext(os.path.basename(tflite_file))[0]
    output_file = f"{base_name}.h"
    array_name = base_name + "_tflite"

    with open(tflite_file, "rb") as f:
        data = f.read()

    with open(output_file, "w") as f:
        f.write(f"unsigned char {array_name}[] = {{\n")
        for i, byte in enumerate(data):
            f.write(f"0x{byte:02x}, ")
            if (i + 1) % 12 == 0:  # Break line every 12 bytes
                f.write("\n")
        f.write("};\n\n")

        f.write(f"unsigned int {array_name}_len = {len(data)};\n")

    print(f"Header file generated: {output_file}")
    print(f"Array length: {len(data)} bytes")

convert_tflite_to_h("trained_models/wakeword/tflite_stream_state_internal_quant/stream_state_internal_quant.tflite")
`
```
I hope you can help me found out what went wrong here. Thank you.

puddly, 2024-12-30:
This was fixed by https://github.com/kahrendt/microWakeWord/pull/39. In the notebook, checkout the main branch of the Git repo instead of the `november-update` branch, it'll include the fix.

bibidentuhanoi, 2024-12-31:
Hi, thanks for your response! I'm using the latest version of the main branch for training. In my most recent notebook (from about 2–3 days ago), I ran: `!git clone -b main https://github.com/kahrendt/microWakeWord` to ensure I'm on the main branch. The model’s output is correctly integrated into the ESPHome code, but I’m encountering an issue: it continuously detects the wake word—even when no one is speaking or there is no background noise.
`Wake word detected!
I (161454) micro_wake_word: Actual tensor arena size is 25796
Wake word detected!
I (162254) micro_wake_word: Actual tensor arena size is 25796
Wake word detected!
I (163034) micro_wake_word: Actual tensor arena size is 25796
Wake word detected!
I (163824) micro_wake_word: Actual tensor arena size is 25796
Wake word detected!
I (164624) micro_wake_word: Actual tensor arena size is 25796`

I'm not directly using a ESPHOME but I'm using this repo  https://github.com/0xD34D/micro_wake_word_standalone to add this wonderful feature into my app. 

Any ideas what went wrong ? Thank you.

kahrendt, 2025-01-21:
I would first try it running on ESPHome and see if it still has the same behavior. I'm not familiar with that project (looks cool though!), so there could be a bug in that implementation.