OHF-Voice/micro-wake-word #53: Request for non-quantized pre-trained models

yf-fan-org, 2025-01-17:
Is it possible to provide a non-quantitative model, because I've found that sometimes it doesn't work well in real-world testing.

yf-fan-org, 2025-01-17:
My native language is not English, and sometimes it's hard to detect when the pronunciation is not quite right, and I wondered if I could improve it by adding some additional audio data. Thank you very much for your answer!
@kahrendt

Nold360, 2025-01-17:
> My native language is not English, and sometimes it's hard to detect when the pronunciation is not quite right, and I wondered if I could improve it by adding some additional audio data. Thank you very much for your answer! [@kahrendt](https://github.com/kahrendt)

I wonder if fine-tuning the models would be possible.. 🤔 maybe that would help?
https://github.com/kahrendt/microWakeWord/issues/38

yf-fan-org, 2025-01-19:
> I wonder if fine-tuning the models would be possible.. 🤔 maybe that would help? [#38](https://github.com/kahrendt/microWakeWord/issues/38)

Thanks, that's a workable approach. Currently I have a new problem, the model architecture I trained using basic_training_notebook.ipynb is different from the given v1 and v2 (from https://github.com/esphome/micro-wake-word-models/tree/main/models) Do you know why?

closed 2025-01-20 by yf-fan-org

kahrendt, 2025-01-21:
In my benchmarking tests, the quantization doesn't reduce the model accuracy in a consistent way. Sometimes it performs slightly better than the floating point model and sometimes slightly worse. When I mean slightly different, I mean the probability cutoff to only have a 5% false rejection rate differs from by 0.01.

There isn't an easy way to run the non-quantized models; they won't work on an ESP32. They really only work using the test functions in the mWW code itself.