OHF-Voice/micro-wake-word #55: Request complete training_notebook.ipynb for v2 model training

yf-fan-org, 2025-01-20:
@kahrendt Hi, I tested training_notebook.ipynb with tag 2.1 but it doesn't work, can you give a full ipynb, thanks a lot!

yf-fan-org, 2025-01-20:
I use the Master branch with the following training command:
“python -m microwakeword.model_train_eval --training_config='training_parameters.yaml' --train 1 --restore_checkpoint 1 --test_tf_nonstreaming 0 --test_tflite_nonstreaming 0 --test_tflite_streaming 0 --test_tflite_streaming_quantized 1 --use_weights "last_weights" inception --cnn1_filters '32' --cnn1_kernel_sizes '5' --cnn1_subspectral_groups '1' --cnn2_filters1 '24,24,24' --cnn2_filters2 '32,64,96' --cnn2_kernel_sizes '3,5,5' --cnn2_subspectral_groups '1,1,1' --cnn2_dilation '1,1,1' --dropout 0.8”

, and finally get the input of the model as 1x1x40, not 1x3x40 as in v2, do I need to change any parameter?

kahrendt, 2025-01-21:
The version 2.1 tag refers to the model version, not the mWW version (this is confusing... I should make the distinction more clear). Try using [basic_training_notebook.ipynb](https://github.com/kahrendt/microWakeWord/blob/main/notebooks/basic_training_notebook.ipynb) in the main repo branch. You are trying to train with the old Inception architecture, but the that notebook uses the newer, more accurate Mixednet architecture.

yf-fan-org, 2025-01-21:
> The version 2.1 tag refers to the model version, not the mWW version (this is confusing... I should make the distinction more clear). Try using [...]

Okay, thanks a lot. One more question, the model I trained with default parameters is not as accurate as the pre-trained model you published, do we need to debug this ourselves? Or can you also give a training configuration reference?