OHF-Voice/micro-wake-word #62: Training notebook no longer works in Google Colab due to upgraded Python to 3.11.11

TurboTechieSystems, 2025-02-20:
Tried running the latest training notebook on Google Colab and first step fails because of Google Colab's upgraded Python version to 3.11.11.  Not sure if there are any breaking changes in this upgrade or not.  It appears the default fallback version is also no longer available on Colab.

First step log is below.

Collecting git+https://github.com/whatsnowplaying/audio-metadata@d4ebb238e6a401bb1a5aaaac60c9e2b3cb30929f
  Cloning https://github.com/whatsnowplaying/audio-metadata (to revision d4ebb238e6a401bb1a5aaaac60c9e2b3cb30929f) to /tmp/pip-req-build-xsbzx0rz
  Running command git clone --filter=blob:none --quiet https://github.com/whatsnowplaying/audio-metadata /tmp/pip-req-build-xsbzx0rz
  Running command git rev-parse -q --verify 'sha^d4ebb238e6a401bb1a5aaaac60c9e2b3cb30929f'
  Running command git fetch -q https://github.com/whatsnowplaying/audio-metadata d4ebb238e6a401bb1a5aaaac60c9e2b3cb30929f
  Running command git checkout -q d4ebb238e6a401bb1a5aaaac60c9e2b3cb30929f
  Resolved https://github.com/whatsnowplaying/audio-metadata to commit d4ebb238e6a401bb1a5aaaac60c9e2b3cb30929f
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: attrs>=18.2 in /usr/local/lib/python3.11/dist-packages (from audio-metadata==0.12.0) (25.1.0)
Collecting bidict==0.* (from audio-metadata==0.12.0)
  Downloading bidict-0.23.1-py3-none-any.whl.metadata (8.7 kB)
Collecting bitstruct<9.0,>=6.0 (from audio-metadata==0.12.0)
  Downloading bitstruct-8.19.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (7.4 kB)
Collecting more-itertools<9.0,>=4.0 (from audio-metadata==0.12.0)
  Downloading more_itertools-8.14.0-py3-none-any.whl.metadata (30 kB)
Collecting pprintpp==0.* (from audio-metadata==0.12.0)
  Downloading pprintpp-0.4.0-py2.py3-none-any.whl.metadata (7.9 kB)
Requirement already satisfied: wrapt<2.0,>=1.0 in /usr/local/lib/python3.11/dist-packages (from audio-metadata==0.12.0) (1.17.2)
Downloading bidict-0.23.1-py3-none-any.whl (32 kB)
Downloading pprintpp-0.4.0-py2.py3-none-any.whl (16 kB)
Downloading bitstruct-8.19.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (82 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 82.9/82.9 kB 7.9 MB/s eta 0:00:00
Downloading more_itertools-8.14.0-py3-none-any.whl (52 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 52.2/52.2 kB 4.8 MB/s eta 0:00:00
Building wheels for collected packages: audio-metadata
  Building wheel for audio-metadata (pyproject.toml) ... done
  Created wheel for audio-metadata: filename=audio_metadata-0.12.0-py3-none-any.whl size=53139 sha256=575a17780a3ed8c3b9703fa67dc35eacd0cbe35ee5bd077924f492ab5f88813d
  Stored in directory: /root/.cache/pip/wheels/f6/63/dd/4a5b3647219e673100fbb14b6c21033c02a6f0a4203e6c2892
Successfully built audio-metadata
Installing collected packages: pprintpp, more-itertools, bitstruct, bidict, audio-metadata
  Attempting uninstall: more-itertools
    Found existing installation: more-itertools 10.6.0
    Uninstalling more-itertools-10.6.0:
      Successfully uninstalled more-itertools-10.6.0
Successfully installed audio-metadata-0.12.0 bidict-0.23.1 bitstruct-8.19.0 more-itertools-8.14.0 pprintpp-0.4.0
Cloning into 'microWakeWord'...
remote: Enumerating objects: 935, done.
remote: Counting objects: 100% (189/189), done.
remote: Compressing objects: 100% (54/54), done.
remote: Total 935 (delta 157), reused 137 (delta 135), pack-reused 746 (from 2)
Receiving objects: 100% (935/935), 31.59 MiB | 10.03 MiB/s, done.
Resolving deltas: 100% (596/596), done.
Obtaining file:///content/microWakeWord
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting audiomentations (from microwakeword==0.1.0)
  Downloading audiomentations-0.39.0-py3-none-any.whl.metadata (11 kB)
Requirement already satisfied: audio_metadata in /usr/local/lib/python3.11/dist-packages (from microwakeword==0.1.0) (0.12.0)
Collecting datasets (from microwakeword==0.1.0)
  Downloading datasets-3.3.1-py3-none-any.whl.metadata (19 kB)
Collecting mmap_ninja (from microwakeword==0.1.0)
  Downloading mmap_ninja-0.7.4-py3-none-any.whl.metadata (13 kB)
Requirement already satisfied: numpy in /usr/local/lib/python3.11/dist-packages (from microwakeword==0.1.0) (1.26.4)
Collecting pymicro-features (from microwakeword==0.1.0)
  Downloading pymicro_features-1.0.0.tar.gz (52 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 52.7/52.7 kB 5.5 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: pyyaml in /usr/local/lib/python3.11/dist-packages (from microwakeword==0.1.0) (6.0.2)
Requirement already satisfied: tensorflow>=2.16 in /usr/local/lib/python3.11/dist-packages (from microwakeword==0.1.0) (2.18.0)
Collecting webrtcvad (from microwakeword==0.1.0)
  Downloading webrtcvad-2.0.10.tar.gz (66 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 66.2/66.2 kB 6.7 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
INFO: pip is looking at multiple versions of microwakeword to determine which version is compatible with other requirements. This could take a while.
ERROR: Package 'microwakeword' requires a different Python: 3.11.11 not in '<3.11,>=3.10'

LaiThePhong, 2025-02-27:
Same issue. But after I changed only `python_requires=">=3.10"` in `setup.py`, It was running successfully in my collab notebook.

XxSimpleAceXx, 2025-03-14:
@drthanwho in google colab, you'll go to the left side of the page, click the folder and browse to microwakeword, then setup.py, and there you can edit the file to the above

disconn3ct, 2025-03-14:
Building on that, you can insert the '!sed' line as shown before you run it and it will go through without errors:
```
!git clone https://github.com/kahrendt/microWakeWord
!sed -i -e 's/">=3.10, <3.11",/">=3.10",/' microWakeWord/setup.py
!pip install -e ./microWakeWord
```

Subcode, 2025-03-26:
Doing this results in:
`ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
torch 2.6.0+cu124 requires nvidia-cublas-cu12==12.4.5.8; platform_system == "Linux" and platform_machine == "x86_64", but you have nvidia-cublas-cu12 12.5.3.2 which is incompatible.
torch 2.6.0+cu124 requires nvidia-cuda-cupti-cu12==12.4.127; platform_system == "Linux" and platform_machine == "x86_64", but you have nvidia-cuda-cupti-cu12 12.5.82 which is incompatible.
torch 2.6.0+cu124 requires nvidia-cuda-nvrtc-cu12==12.4.127; platform_system == "Linux" and platform_machine == "x86_64", but you have nvidia-cuda-nvrtc-cu12 12.5.82 which is incompatible.
torch 2.6.0+cu124 requires nvidia-cuda-runtime-cu12==12.4.127; platform_system == "Linux" and platform_machine == "x86_64", but you have nvidia-cuda-runtime-cu12 12.5.82 which is incompatible.
torch 2.6.0+cu124 requires nvidia-cudnn-cu12==9.1.0.70; platform_system == "Linux" and platform_machine == "x86_64", but you have nvidia-cudnn-cu12 9.3.0.75 which is incompatible.
torch 2.6.0+cu124 requires nvidia-cufft-cu12==11.2.1.3; platform_system == "Linux" and platform_machine == "x86_64", but you have nvidia-cufft-cu12 11.2.3.61 which is incompatible.
torch 2.6.0+cu124 requires nvidia-curand-cu12==10.3.5.147; platform_system == "Linux" and platform_machine == "x86_64", but you have nvidia-curand-cu12 10.3.6.82 which is incompatible.
torch 2.6.0+cu124 requires nvidia-cusolver-cu12==11.6.1.9; platform_system == "Linux" and platform_machine == "x86_64", but you have nvidia-cusolver-cu12 11.6.3.83 which is incompatible.
torch 2.6.0+cu124 requires nvidia-cusparse-cu12==12.3.1.170; platform_system == "Linux" and platform_machine == "x86_64", but you have nvidia-cusparse-cu12 12.5.1.3 which is incompatible.
torch 2.6.0+cu124 requires nvidia-nvjitlink-cu12==12.4.127; platform_system == "Linux" and platform_machine == "x86_64", but you have nvidia-nvjitlink-cu12 12.5.82 which is incompatible.
gcsfs 2025.3.0 requires fsspec==2025.3.0, but you have fsspec 2024.12.0 which is incompatible.`

followed by:

`DEBUG:__main__:Loading /content/piper-sample-generator/models/en_US-libritts_r-medium.pt
Traceback (most recent call last):
  File "/content/piper-sample-generator/generate_samples.py", line 481, in <module>
    main()
  File "/content/piper-sample-generator/generate_samples.py", line 477, in main
    generate_samples(**args)
  File "/content/piper-sample-generator/generate_samples.py", line 76, in generate_samples
    torch_model = torch.load(model_path)
                  ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/torch/serialization.py", line 1470, in load
    raise pickle.UnpicklingError(_get_wo_message(str(e))) from None
_pickle.UnpicklingError: Weights only load failed. This file can still be loaded, to do so you have two options, do those steps only if you trust the source of the checkpoint. 
	(1) In PyTorch 2.6, we changed the default value of the `weights_only` argument in `torch.load` from `False` to `True`. Re-running `torch.load` with `weights_only` set to `False` will likely succeed, but it can result in arbitrary code execution. Do it only if you got the file from a trusted source.
	(2) Alternatively, to load with `weights_only=True` please check the recommended steps in the following error message.
	WeightsUnpickler error: Unsupported global: GLOBAL piper_train.vits.models.SynthesizerTrn was not an allowed global by default. Please use `torch.serialization.add_safe_globals([SynthesizerTrn])` or the `torch.serialization.safe_globals([SynthesizerTrn])` context manager to allowlist this global if you trust this class/function.

Check the documentation of torch.load to learn more about types accepted by default with weights_only https://pytorch.org/docs/stable/generated/torch.load.html.

---------------------------------------------------------------------------

ValueError                                Traceback (most recent call last)

[<ipython-input-2-3f813e085ee0>](https://localhost:8080/#) in <cell line: 0>()
     25 get_ipython().system('python3 piper-sample-generator/generate_samples.py "{target_word}"  --max-samples 1  --batch-size 1  --output-dir generated_samples')
     26 
---> 27 Audio("generated_samples/0.wav", autoplay=True)

[/usr/local/lib/python3.11/dist-packages/IPython/lib/display.py](https://localhost:8080/#) in __init__(self, data, filename, url, embed, rate, autoplay, normalize, element_id)
    114         if self.data is not None and not isinstance(self.data, bytes):
    115             if rate is None:
--> 116                 raise ValueError("rate must be specified when data is a numpy array or list of audio samples.")
    117             self.data = Audio._make_wav(data, rate, normalize)
    118 

ValueError: rate must be specified when data is a numpy array or list of audio samples.`

aegjoyce, 2025-04-27:
See my comments in https://github.com/kahrendt/microWakeWord/issues/65 in order to fix the `torch.load` issue:

> Open the notebook and run step 1. After this has completed you will need to edit piper-sample-generator/generate_samples.py and change the line `'torch_model = torch.load(model_path)` to `torch_model = torch.load(model_path, weights_only=False)`