OHF-Voice/micro-wake-word #63: Is there inference code in C?

lhleohuang, 2025-03-16:
I wanted to test my model on a microcontroller, but it only works with C. [pymicro-wakeword](https://github.com/OHF-Voice/pymicro-wakeword) is written in python, and has several python dependencies. Thanks!

kahrendt, 2025-03-18:
I don't have a plain C implementation, but my original implementation for ESPHome is written in C++: <https://github.com/esphome/esphome/tree/dev/esphome/components/micro_wake_word> 

That code is built (but heavily modified) on the TensorFlow Lite Micro MicroSpeech example:
<https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech>

If you build one a more standalone version, please share!

closed 2025-03-18 by kahrendt

lhleohuang, 2025-03-18:
Thanks - I'll look into that!